Nossiac/mtk-openwrt-feeds #106: Please update drivers

budetisufu, 2019-04-13:
Hi!
The current driver is not suitable for use with 18.06.2. Could you update the drivers?

michallipka, 2019-04-17:
It is suitable. Download the latest available driver for 4.14 kernel (so 4.14.93 driver) and it works on openwrt 18.06.2

budetisufu, 2019-04-18:
I have almost no experience in this matter. I would be grateful for your help.
```
scp mt7628-for-mt76x8-linux-4.14.93.ko root@192.168.1.1:/tmp/mt7628.ko
root@OpenWrt:~# ls /lib/modules/4.14.111/ | grep mt
mt7628.ko
root@OpenWrt:~# modprobe mt7628
failed to mmap /lib/modules/4.14.111/mt7628.ko
root@OpenWrt:~#
```

michallipka, 2019-04-19:
If you have no experience I would suggest to install latest openwrt snapshot and use open drivers present there - They are stable enough.
If you insist on trying closed drivers check issue #104 - you have full walkthrough there

budetisufu, 2019-04-19:
The open source driver still works poorly.
I did everything right, but it turned out that the driver located in /lib/modules/<your-linux-version>/ doesn’t weigh anything, it’s empty. And it is strange.

michallipka, 2019-04-20:
So probably you haven't done EVERYTHING right ;)
there are no compiled drivers for kernel you are trying to compile. You have to use latest available at http://nossiac.com/download/mtk-wifi-ko/ (4.14.93 for kernel 4.14.x).
to have them automatically included in build please edit Makefile(s) <buildroot>/feeds/mtk/drivers/mt76{03,12}/Makefile and change line PKG_KO:=$(PKG_NAME)-for-$(CONFIG_TARGET_SUBTARGET)-linux-$(LINUX_VERSION).ko   - replace $(LINUX_VERSION) with 4.14.93 and you should be good.
you can also download those drivers separately and just place them as files wherever you want and insmod them.

EUA, 2019-05-08:
> The open source driver still works poorly.
> I did everything right, but it turned out that the driver located in /lib/modules// doesn’t weigh anything, it’s empty. And it is strange.

I just wondering what is the issue with open source driver? Thx.

budetisufu, 2019-05-10:
> I just wondering what is the issue with open source driver? Thx.

It's all about speed. When using a proprietary driver, the speed is several times higher.