Nossiac/mtk-openwrt-feeds #127: drivers for mt7613

bricco1981, 2022-12-29:
hi, thanks a lot for your help with this drivers. I'm try to understand, to get drivers for mt7613be chipset, but I cannot find any info. I'm talking about mtk and not mt76.should this card work with mt7615?

vikingloudmouth, 2023-10-15:
any updates on this?
Did you end up working this out @bricco1981

bricco1981, 2023-10-15:
@vikingloudmouth no, i get alot of errors, i stopped now. maybe i''ll start again when i 'll have free time. have you dome progress?

vikingloudmouth, 2023-10-15:
I have tried to use the mt7515 driver. I was able to build the driver with openwrt 22.03.5 and `running` modprobe on the .ko runs without errors. I can't bring it up though. Using Luci I get the message "Please restart manually to create this interface." when selecting enable - this doesn't change after rebooting the router or pressing the reload button. Looking in the log I have errors 

`daemon.err uhttpd[1488]: cat: can't open '/tmp/mtk/if-save': No such file or directory`
` cat: can't open '/tmp/mtk/vifup-save': No such file or directory`

I then tried the command line tool `/sbin/mtkwifi`, first I got a permission error trying to run it. After fixing that with chmod, I can see the same errors that the luci interface generates.

I also tried manually bringing the interface up with 
`ifconfig rai0 up`
but that gives me the error
`ifconfig: SIOCGIFFLAGS: No such device`

From a quick look at the code for the luci app, it appears to me to all start from the what is returned when running
`ls /sys/class/net`

For me, I can see `ra0` in the output of that command ( I was also trying - without issues - the mt7603 driver), but no `rai0` is listed. This is where I got stuck though.

bricco1981, 2023-10-15:
No,mt7615 are not the driver you Need . You Need mt7663.

vikingloudmouth, 2023-10-15:
@bricco1981 Do you know where they are available? I thought the mt7615 driver covered that? (or is that just with the opensource mt76 driver version)

vikingloudmouth, 2023-11-05:
I have been using the mt7663 driver on my mt7613 without any issues for the last week. It was built based on "MT7663_LinuxAP_6.0.3.0_20200106-775e6a.tar" with modifications copied from the mt7615d driver in https://github.com/Azexios/openwrt-r3p-mtk to get it to compile for openwrt 22.03.5.

Does anyone know what need to be changed to use these mtk drivers for use with openwrt 23.05?

bricco1981, 2023-11-06:
so can you share the driver on github? or explain neded modification?

ravichaudhary133, 2024-01-01:
@vikingloudmouth Hey, could you share the process of building the mtk drivers based on mt7663 for mt7613 device. I really need to get it working for my device. The default mt76 driver in openwrt has tx power issues.

bricco1981, 2024-05-02:
@vikingloudmouth can you please share the code, i'm not able to make this work.

vikingloudmouth, 2024-05-08:
I'll share what I have a chance, should find time to dig it back up gain and work out what I did in the next few weeks.

I've been using it since I first posted with only one issue: intermittent reconnection problems when using clients only capable of WPA2 if the router is set to WPA2/WPA3 mixed mode. No issues with clients capable of WPA3, and no issues with WPA2 clients if I don't use the mixed mode and instead only set WPA2 mode. I also have tested it on openwrt 23.05 and the wireless worked fine - the config page wouldn't load though. Hopefully someone else can fix that after I share the code

bricco1981, 2024-05-08:
ok do not worry about config page i can make that easy. i'll make luci app mtk custom for this.

bricco1981, 2024-05-28:
FATAL: modpost: GPL-incompatible module mt_wifi.ko uses GPL-only symbol '__kernel_write'
how solve this?

bricco1981, 2024-06-12:
ping @vikingloudmouth

bricco1981, 2024-07-02:
well as i said i did a custom luci app for this, i do not if you will share this.
[image]
[image]

vikingloudmouth, 2024-07-03:
Sorry for the lack of response. Left all the code in a mess and haven't got around to working out which of them actually works.  Give [mt7663.ko](https://mega.nz/file/oIx3iSIZ#NAt0wHp-Ha1YTJL507kbwKCmLo4AX26FZBMA9A0-rxw) with OpenWrt 23.05 a go. I think it should work with
```
modprobe mt7663
ifconfig rai0 up
brctl addif br-lan rai0
```
and looks for a `/etc/wireless/mt7663/mt7663.2.dat` for the config file. Let me know how it goes. It may give only give very slows speeds though, I may need to work out again how I pointed it at the right data for the radio config. 

I'll try to find the code that that actually can from. I really should have done this properly from the start...

bricco1981, 2024-07-03:
this is crazy i made the driver work but very low signal, so calibration is not correct. i'll try your mt7663.ko

vikingloudmouth, 2024-07-03:
From memory the part of the code that looks for the firmware .bin file was kinda broken when the radio config was for two cards w.r.t. to using/not using a offset into the calibration area for the second radio in a .bin file was used. Think I ended up working around it be pointing it to a .bin file that contained the calibration data for both radios, setting a offset somewhere in the menuconfig, and making sure that I had it set as the second card with the first card selected as a 7603 (even though I wasn't using the mtk driver for the first card)

Think I also had speed issues at one point if I was running with a default .dat without encryption (or something along those lines) even I using no password on the network

vikingloudmouth, 2024-07-03:
If helpful, this is the relevant part of my config
```
CONFIG_PACKAGE_kmod-mt7603=y
# CONFIG_PACKAGE_kmod-mt7603e is not set
# CONFIG_PACKAGE_kmod-mt7615-firmware is not set
# CONFIG_PACKAGE_kmod-mt7615d is not set
# CONFIG_PACKAGE_kmod-mt7615e is not set
# CONFIG_PACKAGE_kmod-mt7663-firmware-ap is not set
# CONFIG_PACKAGE_kmod-mt7663-firmware-sta is not set
CONFIG_PACKAGE_kmod-mt7663e=y
CONFIG_MTK_MT7663E_SUPPORT_OPENWRT=y
CONFIG_MTK_MT7663E_WIFI_DRIVER=y
# CONFIG_MTK_MT7663E_FIRST_IF_NONE is not set
CONFIG_MTK_MT7663E_FIRST_IF_MT7603E=y
# CONFIG_MTK_MT7663E_FIRST_IF_MT7615E is not set
# CONFIG_MTK_MT7663E_FIRST_IF_MT7622 is not set
# CONFIG_MTK_MT7663E_SECOND_IF_NONE is not set
CONFIG_MTK_MT7663E_SECOND_IF_MT7663E=y
# CONFIG_MTK_MT7663E_SECOND_IF_MT7615E is not set
# CONFIG_MTK_MT7663E_SECOND_IF_MTTEST is not set
CONFIG_MTK_MT7663E_THIRD_IF_NONE=y
# CONFIG_MTK_MT7663E_THIRD_IF_MT7615E is not set
CONFIG_MTK_MT7663E_RT_FIRST_CARD=7603
CONFIG_MTK_MT7663E_RT_SECOND_CARD=7663
CONFIG_MTK_MT7663E_RT_FIRST_IF_RF_OFFSET=0x0
CONFIG_MTK_MT7663E_RT_SECOND_IF_RF_OFFSET=0x8000
CONFIG_MTK_MT7663E_MT_WIFI=y
CONFIG_MTK_MT7663E_MT_WIFI_PATH="mt_wifi"

#
# WiFi Generic Feature Options
#
CONFIG_MTK_MT7663E_FIRST_IF_EEPROM_FLASH=y
# CONFIG_MTK_MT7663E_FIRST_IF_EEPROM_EFUSE is not set
CONFIG_MTK_MT7663E_RT_FIRST_CARD_EEPROM="flash"
CONFIG_MTK_MT7663E_SECOND_IF_EEPROM_FLASH=y
# CONFIG_MTK_MT7663E_SECOND_IF_EEPROM_PROM is not set
# CONFIG_MTK_MT7663E_SECOND_IF_EEPROM_EFUSE is not set
CONFIG_MTK_MT7663E_RT_SECOND_CARD_EEPROM="flash"
CONFIG_MTK_MT7663E_MULTI_INF_SUPPORT=y
CONFIG_MTK_MT7663E_WIFI_BASIC_FUNC=y
CONFIG_MTK_MT7663E_DOT11_N_SUPPORT=y
CONFIG_MTK_MT7663E_DOT11_VHT_AC=y
CONFIG_MTK_MT7663E_G_BAND_256QAM_SUPPORT=y
CONFIG_MTK_MT7663E_TPC_SUPPORT=y
# CONFIG_MTK_MT7663E_THERMAL_PROTECT_SUPPORT is not set
CONFIG_MTK_MT7663E_ICAP_SUPPORT=y
CONFIG_MTK_MT7663E_SPECTRUM_SUPPORT=y
CONFIG_MTK_MT7663E_BACKGROUND_SCAN_SUPPORT=y
CONFIG_MTK_MT7663E_SMART_CARRIER_SENSE_SUPPORT=y
# CONFIG_MTK_MT7663E_SCS_FW_OFFLOAD is not set
CONFIG_MTK_MT7663E_MT_DFS_SUPPORT=y
# CONFIG_MTK_MT7663E_OFFCHANNEL_SCAN_FEATURE is not set
CONFIG_MTK_MT7663E_HDR_TRANS_TX_SUPPORT=y
CONFIG_MTK_MT7663E_HDR_TRANS_RX_SUPPORT=y
CONFIG_MTK_MT7663E_WSC_INCLUDED=y
CONFIG_MTK_MT7663E_WSC_V2_SUPPORT=y
CONFIG_MTK_MT7663E_DOT11W_PMF_SUPPORT=y
CONFIG_MTK_MT7663E_TXBF_SUPPORT=y
# CONFIG_MTK_MT7663E_FTM_SUPPORT is not set
# CONFIG_MTK_MT7663E_MBO_SUPPORT is not set
CONFIG_MTK_MT7663E_IGMP_SNOOP_SUPPORT=y
CONFIG_MTK_MT7663E_RTMP_FLASH_SUPPORT=y
CONFIG_MTK_MT7663E_CAL_BIN_FILE_SUPPORT=y
CONFIG_MTK_MT7663E_LINK_TEST_SUPPORT=y
CONFIG_MTK_MT7663E_WIFI_GPIO_CTRL=y
# CONFIG_MTK_MT7663E_MIN_PHY_RATE_SUPPORT is not set
# CONFIG_MTK_MT7663E_VENDOR_FEATURE11_SUPPORT is not set
CONFIG_MTK_MT7663E_ATE_SUPPORT=y
# CONFIG_MTK_MT7663E_MEMORY_OPTIMIZATION is not set
# CONFIG_MTK_MT7663E_PASSPOINT_R2 is not set
CONFIG_MTK_MT7663E_UAPSD=y
# CONFIG_MTK_MT7663E_TCP_RACK_SUPPORT is not set
CONFIG_MTK_MT7663E_RED_SUPPORT=y
# CONFIG_MTK_MT7663E_FDB_SUPPORT is not set
# CONFIG_MTK_MT7663E_FIRST_IF_EPAELNA is not set
CONFIG_MTK_MT7663E_FIRST_IF_IPAILNA=y
# CONFIG_MTK_MT7663E_FIRST_IF_IPAELNA is not set
# CONFIG_MTK_MT7663E_SECOND_IF_EPAELNA is not set
CONFIG_MTK_MT7663E_SECOND_IF_IPAILNA=y
# CONFIG_MTK_MT7663E_SECOND_IF_IPAELNA is not set
# CONFIG_MTK_MT7663E_RLT_MAC is not set
# CONFIG_MTK_MT7663E_RTMP_MAC is not set
# end of WiFi Generic Feature Options
```

I am using a .bin file I extracted from by device. not sure exactly what I did to extract it, but the radio calibration data is at offset `0x8000` that matches the above config. You can tell by the first two bytes at that offset being `0x63 0x76` which matches `7663`. The first part of my .bin file is for the 7603 radio, so at offset `0` the first two bytes are `0x03 0x76` for `7603`.

Hopefully, this help you getting it to find thee calibration data

bricco1981, 2024-07-03:
so  when you first fixed the compilation, did the driver worked fine? good speed and signal? or always week?

vikingloudmouth, 2024-07-04:
No I had speed issues initially. Don't think I needed to change the code to fix it though, just the config and point the driver to a .bin file that contained the calibration data at the offset rather than at the start.

From memory, the driver was putting out some messages in one of openwrt logs (think it was the kernel log) about not finding firmware/eeprom/bin data and using a default one.

Also pretty sure I've found the code that works. I'll give it a quick test from a clean start and put it up if it works.

vikingloudmouth, 2024-07-04:
> FATAL: modpost: GPL-incompatible module mt_wifi.ko uses GPL-only symbol '__kernel_write'
> how solve this?
> what does this error mean

You probably already worked this out, but I got around this by changing the mt_wifi code to say it was GPL...

bricco1981, 2024-07-04:
yes i i fixed that. the problem that now i have is wrong macaddress and weak signal in ap mode, in sta mode i get good speed like 300 down 50 up.

vikingloudmouth, 2024-07-04:
If your getting good speeds in sta mode that means you now have it finding the calibration data right?

Try checking your config in AP mode, think I made have also had slows speeds until I setup the .dat file to use encryption (or something along those lines) for the network mode.  From memory it wasn't really a problem with the driver, just that devices would only use a slow rate in the outdated default mode without any

bricco1981, 2024-07-04:
well the signal is too weak in ap mode and why i have wrong mac address?

vikingloudmouth, 2024-07-04:
I have no idea about the mac address, I didn't check what it was before and don't know if it is wrong or not for me

bricco1981, 2024-07-04:
could be that the antennas are not used the right way. my device is a comfast ew72 v2

vikingloudmouth, 2024-07-04:
Doesn't the good speed in STA mode rule that out?

bricco1981, 2024-07-04:
yes that's very strange. also i do not see about ANTENNA STAFF in bin folder, if you check the bin folder mt7615 you can see antenna things.

vikingloudmouth, 2024-07-04:
you mean the .bin files?

bricco1981, 2024-07-04:
yes IPAELNA ect

vikingloudmouth, 2024-07-04:
Pretty sure I'm just running with one .bin file. I extracted it from openwrt using the button to save the radio mtdblock, Not sure if it's needed in both places, but I called the file `mt7663_7603.bin` and put a copy into `/lib/firmware/mt7663_7603.bin` and `/lib/firmware/mt7663_7603.bin` . Pretty sure that is what the .ko I uploaded looks for

bricco1981, 2024-07-04:
strange for me it look in etc/wireless/mt7663e/mt7663.2.eeprom.bin

vikingloudmouth, 2024-07-04:
I may have changed that in the menuconfig. I did initially have issues with it not finding the right file / offset into the file. 

Can you see error messages in the log about it?

bricco1981, 2024-07-04:
but you have weak signal? or normal?

vikingloudmouth, 2024-07-04:
Signal strength looks fine, laptop says 5 bars. No issues with speed

bricco1981, 2024-07-04:
do you get this in log? 
get_wdev_by_idx: invalid idx(0)

bricco1981, 2024-07-04:
well i'll wait for your code so i can test if i have same weak signal.

bricco1981, 2024-07-04:
that's the app custom, to make the clients show right, make a folder in etc/wireless/mt7612
[luci-app-mtwifi.zip](https://github.com/user-attachments/files/16096168/luci-app-mtwifi.zip)
also i think is good if you edit the name of your .ko  mt_wi.ko in the first makefile and in makefile ap so driver will load and you can enable the radio in luci.

vikingloudmouth, 2024-07-04:
> get_wdev_by_idx: invalid idx(0)

Seem familiar, think I had that when it wasn't finding the .bin file.

bricco1981, 2024-07-04:
well i'm not sure why you .ko is so big, anyway it killed my device.

bricco1981, 2024-07-05:
are you going to share this working code?

bricco1981, 2024-07-05:
Well i hope so, i'm working on 23 too,kernel 5.15.x

vikingloudmouth, 2024-07-06:
@bricco1981  https://github.com/vikingloudmouth/mt7663
The correct firmware file needs to exists for a good signal. By default, it looks for /lib/firmware/mt7663.bin, you can change the file it looks for in the l1 profile options. Alternatively
 `dd if=/dev/$(grep -i '"factory"' /proc/mtd | cut -c 1-4) of=/lib/firmware/mt7663.bin bs=1 count=1536 skip=32768`
should give you the right file for your device.

I've renamed the .ko to `mt_wifi.ko` as you suggested, but couldn't get your luci app to work - won't compile. Getting a warning `has a dependency on 'luci-lib-jquery-1-4', which does not exist`. Tried removing that dependency, it then builds but then I just get a blank page in luci for that page.

bricco1981, 2024-07-06:
hi, same as my code weak signal 
=== pAd = (ptrval), size = 7149568 ===
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.947059]
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.957808] <-- RTMPAllocAdapterBlock, Status=0
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.962351] pAd->PciHif.CSRBaseAddress =0xc0d00000, csr_addr=0xc0d00000!
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.969110] RTMPInitPCIeDevice():device_id=0x7663
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.973807] mt_pci_chip_cfg(): HWVer=0x8b10, FWVer=0x8a01, pAd->ChipID=0x7663
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.980936] mt7663_init()-->
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.983822] <--mt7663_init()
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.991662] <-- RTMPAllocTxRxRingMemory, Status=0
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   11.996386] get_wdev_by_idx: invalid idx(0)
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   12.000701] get_wdev_by_idx: invalid idx(0)
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   12.004969] get_wdev_by_idx: invalid idx(0)
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   12.009165] get_wdev_by_idx: invalid idx(0)
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   12.013338] get_wdev_by_idx: invalid idx(0)
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   12.017519] get_wdev_by_idx: invalid idx(0)
Fri Mar 22 22:11:03 2024 kern.warn kernel: [   12.022997] get_wdev_by_idx: invalid idx(0)

bricco1981, 2024-07-06:
that's for luci mtk
[luci-lib-jquery-1-4_6_mipsel_24kc.zip](https://github.com/user-attachments/files/16115703/luci-lib-jquery-1-4_6_mipsel_24kc.zip)

bricco1981, 2024-07-06:
i cannot go from one room to another is to weak, if signal is good for you, share mt7663.bin

vikingloudmouth, 2024-07-06:
> but still weak, we need fix this

Don't know how to help there. All my issues with low signal strength disappeared as some as I put the .bin in the `/lib/firmware/` folder with a name that matches the setting in the l1 profile.

Just checked my log, I'm getting a bunch of `get_wdev_by_idx: invalid idx(0)`, everthing seems to be working fine though...

vikingloudmouth, 2024-07-06:
> share mt7663.bin

No need, just run 
`dd if=/dev/$(grep -i '"factory"' /proc/mtd | cut -c 1-4) of=/lib/firmware/mt7663.bin bs=1 count=1536 skip=32768`

To check it is right, then run
`hexdump /lib/firmware/mt7663.bin`
The first 2 bytes should be 0x7663.

bricco1981, 2024-07-06:
root@OpenWrt:~#  hexdump /lib/firmware/mt7663.bin
0000000 7663 0100 e1e0 6fa9 e458 7663 14c3 8000
0000010 0002 7663 14c3 0000 0000 0000 0000 0000
0000020 0000 0000 0000 0000 0000 0000 0000 00e3
0000030 0100 0080 0000 0000 0000 0000 0000 0000
0000040 0000 0000 0000 0000 0000 8089 9080 0001
0000050 0000 0000 0000 0000 0000 0000 0000 0000
*
0000070 c6c6 c4c4 c1c2 c200 c200 00c1 8100 8584
0000080 c2c2 81c1 8281 8685 c500 c2c5 00c2 00c3
0000090 c1c3 0000 8382 c384 c1c3 0000 8382 0084
00000a0 c3c3 00c1 8200 8483 0000 0000 0000 0000
00000b0 ded1 1b23 1b1b 0d15 0d0d 4040 4040 0000
00000c0 0000 0000 0000 0000 0000 0000 0000 0000
00000d0 0000 0000 0000 0000 0000 0000 0000 3c00
00000e0 4310 29be 0000 0000 0000 0000 0000 0000
00000f0 4310 29bd 0000 0000 0000 0000 0000 0000
0000100 0000 0000 0000 0000 0000 0000 0000 0000
*
0000150 0000 0000 0008 01e0 0000 0000 0000 0000
0000160 0000 0000 0000 0000 0000 0000 0000 0000
0000170 0000 0000 0000 0000 1000 2810 0000 0000
0000180 0000 0000 0000 0000 0000 0000 0000 0000
*
00001b0 1212 0002 0000 0500 0000 0000 0086 0000
00001c0 0000 0000 0000 0000 0000 0000 0000 0000
*
00001f0 d3cc d3cc d3cc d3cc d3cc d3cc d3cc d3cc
0000200 1623 1616 0a17 0a0a 4040 4040 0000 0000
0000210 0000 0000 0000 0000 0000 0000 0000 0000
*
0000230 0000 0000 0000 0000 0000 3a00 0000 0000
0000240 2020 4320 22c7 c743 4322 22c7 c743 c322
0000250 22c7 c7c3 4322 22c7 c6c3 0022 c2c2 c3c2
0000260 c4c4 c4c4 c4c4 c5c5 c6c5 cbc8 0000 0000
0000270 0000 0000 0000 0000 0000 0000 0000 0000
*
00002b0 2020 4320 22c7 c743 4322 22c7 c6c3 4322
00002c0 22c6 c743 4322 22c7 c743 0022 c3c3 c7c4
00002d0 c9c9 cbca cacb c9c9 c9c9 ccca 0000 0000
00002e0 0000 0000 0000 0000 0000 0000 0000 0000
*
0000600
root@OpenWrt:~#

vikingloudmouth, 2024-07-06:
Seems right, first part is 7663 as expected. Slightly different to mine, but mine is also different every other one I found in the gpl code releases for different routers I was searching through. 

Really don't think it will help, but here is mine [mt7663.bin](https://mega.nz/file/xFYgQYRb#GCigcfPmiUg1gJmxzY9fEBrUei3C3FLOtJYreeiUcOI).

I did also see somewhere that suggested issues with radio calibration data can sometimes be resolved by flashing the full original stock firmware back on, and then reflashing openwrt. The idea was to restore the data back to what it should be with the stock firmware incase it had been unintentionally modified somehow.

vikingloudmouth, 2024-07-06:
> that's for luci mtk
> [luci-lib-jquery-1-4_6_mipsel_24kc.zip](https://github.com/user-attachments/files/16115703/luci-lib-jquery-1-4_6_mipsel_24kc.zip)

How do I add this to my build? 

Realise I could just install the package once I have flashed my image, but would prefer to have everything cleanly built and included in the build process.

bricco1981, 2024-07-06:
unpack and add this in builroot   /feeds/routing/
[luci-app-bmx6.zip](https://github.com/user-attachments/files/16115854/luci-app-bmx6.zip)

vikingloudmouth, 2024-07-06:
> and why i have wrong mac address?

Looks like the mac address come from the .bin file https://github.com/Nossiac/mtk-openwrt-feeds/issues/41. Just edit the hex file to be whatever you want

bricco1981, 2024-07-06:
i know that but i do not know why this hell of device have low signal, and the driver eat alot of memory

vikingloudmouth, 2024-07-06:
> device have low signal

I assume that you have checked this is a driver issue and that the stock firmware doesn't have this problem?

bricco1981, 2024-07-06:
i never used stock firmware, just to flash wrt

vikingloudmouth, 2024-07-06:
> i never used stock firmware, just to flash wrt

Might be worth checking, verify it isn't a device/hardware problem giving poor signal.

bricco1981, 2024-07-06:
i do not think so mt76 have better range

bricco1981, 2024-07-06:
have your device built in PA? power amplifier

bricco1981, 2024-07-06:
i have found a way to let load the bin whitout extracting it.
this part is wrong

* @ used mtd mode flash partition from proprietary driver mt_wifi_mtd.c
*/
int mt_mtd_write_nm_wifi(char *name, loff_t to, size_t len, const u_char *buf);
int mt_mtd_read_nm_wifi(char *name, loff_t from, size_t len, u_char *buf);

#define flash_read(_ctrl, _ptr, _offset, _len) mt_mtd_read_nm_wifi("Factory", _offset, (size_t)_len, _ptr)
#define flash_write(_ctrl, _ptr, _offset, _len) mt_mtd_write_nm_wifi("Factory", _offset, (size_t)_len, _ptr)

#else
/*
* @ use sdk export func.
*/

extern int ra_mtd_write_nm(char *name, loff_t to, size_t len, const u_char *buf);
extern int ra_mtd_read_nm(char *name, loff_t from, size_t len, u_char *buf);

#define flash_read(_ctrl, _ptr, _offset, _len) ra_mtd_read_nm("Factory", _offset, (size_t)_len, _ptr)
#define flash_write(_ctrl, _ptr, _offset, _len) ra_mtd_write_nm("Factory", _offset, (size_t)_len, _ptr)

just replacinf Factory with factory in embedded/common/ee_flash.c

[249 more replies]